APEbbers/FreeCAD-Ribbon #301: Workflow for renaming a panel by manipulating the Combine Panel command

raindropsfromsky, 2025-02-10:
Starting a new issue, as the other issue was getting cluttered with unrelated topics...

I tried renaming the panel. 

The interface seems changed. 
However, there are still problems in the workflow:

I completed the two steps (entered a name for the panel, and then added the constituent panelsto the right pane).
When I pressed the **Create Panel** button (1), it changed the first field from "New" to my newly given name for the combined panel.
[image]

This workflow has a few issues:
1. At this stage, my newly entered name for the custom panel is loaded in place of the Select custom panel.
    However, the Left side of panel selection UI still shows the original panel names.
    The original panel name should have been replaced by the new name.

2. At this stage, the UI does not have offer any affordance for the next step.
    What is the user supposed to do?
    Although the manual may say that choosing "new" in the Select custom Panel field will clear all entries, this is not intuitive.
    
3. If the user clicks "remove panel", what will happen to the commands that are assigned to it?
    From the UI, my intference is that they will be deleted.
    But that should not happen. 
    It is preferable to let the original panels remain, and add the custom panels.
    This way, the user can hide the original panels and show the custom panels.

4. I selected Close, thinking that my new custom panel was already created.
    FC restarted.
    But I could not find my newly created custom panel in the list.    
    [image]
    The left panel should show all original panels and custom panels that are previously created.
    Otherwise I cannot load a custom panel and edit it further.

APEbbers, 2025-02-10:
Hi,

I'm already working on it. see #300.

kind regards

closed 2025-02-10 by APEbbers

raindropsfromsky, 2025-02-11:
[image]

I forgot to mention another issue here: 
The left pane shows two instances of the same panel (ExplodedAssembIyAuxiIiarAssembIyTooIs)

**Desired:** 
Show only one instance of each panel.
(There is no need to create a duplicate of any panel, so this must be a bug.)