APEbbers/FreeCAD-Ribbon #306: The default small icons do not have adequate vertical separation: Every icon appears cropped from the bottom side (even the bottom icons)

raindropsfromsky, 2025-02-11:
The default small icons do not have adequate vertical separation: Every icon appears cropped from the bottom side (even the bottom icons)
[image]

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40176 (Git) Conda
Build type: Release
Branch: main
Hash: 2745f436026b6f1fb84dbaaf89ea22b5fb4c2295
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical/physical DPI: 96/123.928
Installed mods: 
  * Beltrami 1.1.0
  * CommandPanel
  * CurvedShapes 1.0.13
  * Curves 0.6.58
  * ExplodedAssembly
  * fasteners 0.5.37
  * frame 0.1.1
  * FrameForge 0.1.2
  * FreeCAD-Ribbon 1.8.0
  * freecad.gears 1.3.0
  * FreeGrid 2.2.0
  * OpenTheme 2024.9.1
  * parts_library
  * Quetzal 1.1.0
  * SearchBar 1.3.3
  * sheetmetal 0.7.10
  * Silk 0.1.5
  * symbols_library

raindropsfromsky, 2025-02-11:
**Update:** 
That problem was with the default values of icon sizes.
I adjusted the icons to 28 (all my icons are "small" and set to the same size.) 

Now the problem is not there.

But probably it is best to check this up.

APEbbers, 2025-02-11:
Thanks, I keep an eye open.

raindropsfromsky, 2025-02-11:
This problem is reproducible: 

1. Change all icons to "small"
2. Reduce the small icon size to 24.

All WBs have the same issue.

The interesting question is, why are buttons that have empty space below are also cropped?
They don't another button (or a text label) to clip them from below.

-------------
Although I found this combination by chance, other combinations also may face the same problem.

APEbbers, 2025-02-11:
Very strange. I followed your steps including the scaling factor and i don't have that problem. It has to do with the calculation of the ribbon height.

Do you also have this when you have large buttons? and medium buttons?

What are the sizes of the other buttons on your system?

raindropsfromsky, 2025-02-11:
> Do you also have this when you have large buttons? and medium buttons?
> What are the sizes of the other buttons on your system?

I was carrying out more experiments, and I found the precise problematic combo.
I have updated my post above with the new information.

APEbbers, 2025-02-11:
> Since all my buttons are small, I never bothered about the size of medium and large buttons.
I varied the tabbar button size, and found that they have no effect on clipping.

The is a function that determines what the largest size is 3x small, 2x medium or 1x large. That's why I'm asking.

APEbbers, 2025-02-11:
It has to do with the large button size. If you set it back to 72, the icons will appear normal.
I'm working on a fix, but this solves the problem in the meantime.

APEbbers, 2025-02-11:
Hi,

The issue is fixed. It was caused because both the small buttons and the large buttons had the same sizes. I updated the algorithm in both v1.7.3.4 and the develop branch.

Thanks for helping me pinpoint the exact conditions! That help a lot. 

kind regards

closed 2025-02-11 by APEbbers

raindropsfromsky, 2025-02-12:
> Thanks for helping me pinpoint the exact conditions! That help a lot.

This is the least I can do, given that I am not a coder.
Please keep up the excellent work! ❤

raindropsfromsky, 2025-02-12:
After we finished our discussion yesterday, I put my laptop to sleep. 
(My timezone is UTC+5.5)

It's possible that my laptop woke up for Windows maintenance tasks, because I see some new errors in the report panel, which appeared while I was sleeping.

They may provide some additional clues, because they are about DPI:

Essentially there are two errors, and both are repeated multiple times.

First, there are about 10 of this:
 `QPixmap::setMask() mask size differs from pixmap size`

Then there are three of this:
`monitorData: Unable to obtain handle for monitor '\\.\DISPLAY1', defaulting to 96 DPI.
`

-----------

I started AM today to check for the updated Ribbon.

I saw the following error in the Report panel:
`Invalid version specified for tag version in Addon FreeCAD Ribbon: 1.8.x`

I have not updated the Ribbon WB yet.

APEbbers, 2025-02-12:
Hi,

> First, there are about 10 of this:
> QPixmap::setMask() mask size differs from pixmap size
> 
> Then there are three of this:
> monitorData: Unable to obtain handle for monitor '\\.\DISPLAY1', defaulting to 96 DPI. 

That might be the cause of the bleeding edges with tabs on your system. I have a look to see if i can find the cause of this.

> I started AM today to check for the updated Ribbon.
> 
> I saw the following error in the Report panel:
> Invalid version specified for tag version in Addon FreeCAD Ribbon: 1.8.x

This error you can ignore. the AM will still update (most of the time). This is only on the develop branch anyway.
When updating from commandline, this is never an issue.

To update via command line: Go the the addon folder. Do a right click on "open in terminal".
Then enter "git pull". That's it!

FYI, I'm on UTC+1, wintertime. I'm from the netherlands. Maybe also good to know.

kind regards

raindropsfromsky, 2025-02-12:
> To update via command line: Go the the addon folder. Do a right click on "open in terminal".
> Then enter "git pull". That's it!

I could not find a "Addon" subfolder in the installation folder. or in C:\Users\naray\AppData\Roaming\FreeCAD.

APEbbers, 2025-02-12:
The next folder is called "mod". 

your path is like: "C:\Users\naray\AppData\Roaming\FreeCAD\mod\FreeCAD-Ribbon".