APEbbers/FreeCAD-Ribbon #308: FC hangs with an error message in Ribbon Design mode

raindropsfromsky, 2025-02-11:
I was trying to cedit the Spreadsheet WB in Ribbon Design section.
FC hanged with the following message:

[image]

This is the text from the **Report** panel:
```
AttributeError: 'NoneType' object has no attribute ' setCheckState '
Traceback (most recent call last) :
File "C: . \LoadDesign Ribbon.py" ,
self. form. CorrmandTab1e RD. item (i 2, 4) . setCheckState (Qt . CheckState . Checked)
AttributeError: 'NoneType' object has no attribute ' setCheckState '
line 2841,
in on TableCe11 RD clicked
```

APEbbers, 2025-02-11:
Hi,

I can confirm. I'm on it

kind regards

APEbbers, 2025-02-11:
Hi,

It is solved in both v1.7.3.3 and the develop branch. It was indeed an sorting issue. Seems an regression from an earlier issue somehow.

kind regards

closed 2025-02-11 by APEbbers