APEbbers/FreeCAD-Ribbon #312: Feature request: Assign shortcuts to specific commands

raindropsfromsky, 2025-02-12:
Let the user assign shortcuts to specific commands.

I know this may not be considered a function of the ribbon UI _per se_.

But since the Ribbon WB completely replaces the original toolbar part of FC, this is the most natural place where users can assign shortcuts.

APEbbers, 2025-02-12:
Hi,

I look into that. But there is an existing menu for that. Go to Tools->Customize. There you should be able to change or add shortcuts.
Maybe I can adopt this menu or create a button for it under preferences.

kind regards

raindropsfromsky, 2025-02-12:
> Maybe I can adopt this menu or create a button for it under preferences.

That's a good idea!

Ideally, the shortcut UI should be embedded inside the **Preferences** dialog.
That gives a better continuity to the user.
For example, add a tab in the Layout window to set hotkeys.

[image]

BTW why have two separate windows for Layout and preferences?
Combine them under one window called "Customize"!

TomPcz, 2025-02-18:
Please also make the shortcuts for Toggle Overlay (currently forced to F4 in the code) and Toggle Transparency (Ctrl+F4) configurable, or ideally, make them respect the settings configured in Tools → Customize. Currently, there is an issue with this function (FreeCAD/FreeCAD#19570), and when the Ribbon is installed, the only way to disable that shortcut is by modifying the code.

APEbbers, 2025-02-18:
Hi,

I can do that no problem. However another solution for in the meantime is to enable FreeCAD's overlay function.
as default, The ribbon replaces FreeCAD's function with a version which excludes the ribbon.
To keep it familiar this function has the same shortcut.

To enable FreeCAD's overlay function, go to Ribbon preferences and check:
[image]

It might be that you have to restart twice before FreeCAD's overlay function is enabled.

Als note that you cannot use both functions together.

kind regards

APEbbers, 2025-02-22:
Hi @TomPcz  and @raindropsfromsky, 

the shortcuts for the overlay function are now following the shortcuts of the original function. Currently this is only in the develop branch. It will be present from version 1.8.0 onwards.

I've set the request for assigning shortcuts from the ribbon for all commands to a feature for version 2.0. With that version, context menus will be implemented.

For now, the customize button is placed under the preference button. (Develop branch)
[image]

kind regards,

closed 2025-02-22 by APEbbers