APEbbers/FreeCAD-Ribbon #314: Suggestion: Please show the Ribbon's Github rev number in Help > About > Ribbon dialog.

raindropsfromsky, 2025-02-12:
When dealing with development branch, the revision number is the best way to exchange information between testers and developer.

However, the **Help > About > Ribbon** dialog does not show this number at all. 
It simply says "1.8.x", which cannot be used to correlate with Git's version control system.
So neither the tester nor the developer can be clear which revision the tester was referring to when reporting a problem.

**Desired:** 
Display the Git revision number in Help > About

APEbbers, 2025-02-12:
Hi,

Agreed. Since last week I've have a function to read from git, so it shouldn't be difficult to implement.

kind regards

APEbbers, 2025-02-12:
Hi,

it is implemented in the develop branch

kind regards

raindropsfromsky, 2025-02-13:
The information does not includes decimal points in the revision number.

[image]

This says 1.8.x, not the specific number.

APEbbers, 2025-02-13:
The 1.8.x is manually created value. the SHA is the full commit ID which is created automatically every commit and is always unique.
[image]

I don't want to update the 1.8.x for every commit that I do, because I make a lot of commits just for testing alone.
Also, 1.8.x indicates which release the develop branch is currently used.

I'll look if i can add a copy button to collect all the details which you can paste. Like FreeCAD itself

kind regards