APEbbers/FreeCAD-Ribbon #317: BUG: The ribbon disappears when a drop-down is selected from a button

raindropsfromsky, 2025-02-14:
Frankly, I am not sure whether this is a bug or a feature!

When we click on the drop-down triangle of a button, the submenu appears, but the ribbon disappears.
[image]

- Some users may see this as a virtue, because we are now focusing on the submenu of the selected button, and we don't need the ribbon.
- Others may feel that the ribbon should not be dismissed because we have not finished the operation.

APEbbers, 2025-02-14:
Hi,

I noticed it too, but I havent found a solution yet. I'm on it though.

Kind regards,

APEbbers, 2025-02-16:
Hi,

This is fixed on the develop branch for both buttons and the overflow menu.

kind regards

closed 2025-03-12 by APEbbers