APEbbers/FreeCAD-Ribbon #318: The 'show on hover' feature should exclude the "FC menu" button

raindropsfromsky, 2025-02-14:
At present, the **show on hover** feature includes the [image] button. In other words, if we hover on this button, the ribbon appears.

[image]

Instead, a hover on [image] should display the menu, like this:

[image]

APEbbers, 2025-02-14:
Hi,

I've implemented this in the develop branch. Now only hovering over the tabbar will show the ribbon. Menu, quickaccess and righttoolbar are excluded.
Also, now when you start FreeCAD, the ribbon is folded by default.

Kind regards,

raindropsfromsky, 2025-02-14:
That's wonderful!

BTW Ribbon's autohide function has never given any issues so far, since you activated it.

APEbbers, 2025-02-14:
Great! I hadn't any issues either, so I'll introduce it with the next release!

kind regards

raindropsfromsky, 2025-02-14:
> I've implemented this in the develop branch. Now only hovering over the tabbar will show the ribbon. Menu, quickaccess and righttoolbar are excluded.

IMHO the solution does not work as required:

The **Menu** button's behavior depends on the direction of approach of the mouse:

1. If the ribbon is visible, and if I move in from the ribbon's area, the **Menu** button does not turn off the ribbon.
    (The **Menu** button still behaves as if it is a part of the ribbon, and does not let the ribbon hide.)   **[BUG]**
2. If the ribbon is NOT visible, and if I bring the mouse on the **Menu** button from below, the **Menu** button does not show the ribbon. **[OK]**

[image]

**Desired:**

1. If the ribbon is visible, and if I move the mouse from the ribbon area to the Menu button, the moment it leaves the ribbon area, turn off the ribbon. When it reaches over the **Menu** button, show the main menu of FC.
2. If the ribbon is NOT visible, and if I hover the mouse on the WB toolbar, show the ribbon. If I hover the mouse on the **Menu** button, show the main menu.  (Thus, there are two hover targets, and two different things to reveal, depending on the target)

APEbbers, 2025-02-14:
Hi,

> (The Menu button still behaves as if it is a part of the ribbon and does not let the ribbon hide.) [BUG]
I'll look into it, but i don't see it as big problem, because as soon as you click on the menu, the ribbon folds anyway.
Actually, it might be a good thing because when you activate the first tab, otherwise you have to move down to prevent the ribbon from folding.

> The Menu button (a hover on it should reveal the main menu)

I disagree. Here you have to click on the menu button like with the dropdown commands. This is common behaviour for software with a ribbon. It also prevents the user from accidently showing the menu when clicking on a command below it.

> The WB toolbar (a hover on it should reveal the ribbon)

I also disagree. (I'm assuming that you mean the toolbar on the right) This should not trigger the ribbon.

I actually implemented like this. You hover over the tabs, when you want to access the ribbon. All other buttons have their own commands and do not need the ribbon to be visible.

kind regards

raindropsfromsky, 2025-02-14:
> Here you have to click on the menu button like with the dropdown commands. This is common behaviour for software with a ribbon. It also prevents the user from accidently showing the menu when clicking on a command below it.

Yes, that's fine too. (I should have added "ideally" to my suggested behavior.)

Here, the problem is that when the mouse comes over the **Menu** button, the ribbon does not hide. 

> I also disagree. (I'm assuming that you mean the toolbar on the right) This should not trigger the ribbon.

I think we are talking about the same thing.
I call the following structure as "WB toolbar":

[image]

APEbbers, 2025-02-14:
Ah I see. the "WB toolbar" is actually called tabbar and the buttons with the workbench icons are called tabs. This is the common terminology for a ribbon or tabbed form. Even FreeCAD calls it tabbar.
[image]
That's the confusion.

However I have thought about activating a workbench on hover in the past, but it will result in sluggish behaviour, because of the loading time of the buttons. especially on older systems.
Besides that, it is also not that common behaviour. SolidWorks, SolidEdge but also MS office don't have this behaviour. 

kind regards

raindropsfromsky, 2025-02-14:
Oh I get it now!

I always turn off the **Quick Access** toolbar, as I can do all those operations with hotkeys.
So it never entered my mind.

No matter! I'd say the Quick Access toolbar also is part of the WB toolbar.
So we need not skirt around it.

To sum up, only the **Menu** button should be excluded from the hot zone.
If the mouse is hovering over any other area, the ribbon should show up.

So the user can be as lazy as he wants! 😄

APEbbers, 2025-02-16:
Hi,

This is now implemented in the develop branch. Including #317 

kind regards

raindropsfromsky, 2025-02-16:
I don't see any difference vis-a-vis earlier version. There is a distinct state-dependency in its behavior:

1. If I move the mouse over the tabbar, the ribbon becomes visible. Now if I move the mouse over the Menu button, the ribbon is still visible. It does NOT turn off. In this condition, we can still click on the **Menu** button to open the menu, but that will not occur to the user because of the ribbon. **[BUG]**

2. If I move the mouse out of the ribbon area, the ribbon hides. Now if I move the mouse to the **Menu** button directly from below **-- WITHOUT letting the mouse touch the tabbar --** the ribbon stays hidden. Now I can click on the **Menu** button to see the manu items.. **[OK]**

Thgus, a part of the problem is still not resolved.

APEbbers, 2025-02-16:
Here it is working:
https://github.com/user-attachments/assets/f5f0e2ec-ba9b-469e-b03c-c9fc0982cfca

perhaps another gitt pull?

raindropsfromsky, 2025-02-17:
> perhaps another gitt pull?

That is quite possible, thanks to the vagaries of the Addon Manager.
First, it always shows the stable version even when I am running 1.8.x
Secondly, if I switch to development branch, it does not pop up the update found dialog.
I have to restart on my own.

I did all of this, and hoped for the best!

BTW this is why I had posted a feature request to publish the Git revision number.
Currently, it is always 1.7.x (stable) or 1.8.x (Development branch).
So I cannot tell _which_ 1.8.x is running!

Imagine it displays a 4-part number such as 1.8.6.1 vs 1.8.7.4.
It will be easy to tell which version I am using.

APEbbers, 2025-02-17:
This is will not happen. The system that you propose is already in the main branch. To do that separatly on the develop branch as well will result in mixups. Besides, you just have to copy the information from the about dialog including the SHA value and i can see which commit you are on. I explained this in issue #314
[image]

Other than that, please use the commandline as i have explained in #306. That works better than the add-on maneger.
The add-on manager works best with main branches.

kind regards

raindropsfromsky, 2025-02-17:
My **About...** page shows:
[image]

I executed a git pull
[image]

To sum up, the git pull does not get me the latest development version.

APEbbers, 2025-02-17:
your on the main branch

APEbbers, 2025-02-17:
what happens if you type "switch Develop"? and then do a git pull?

You can ignore the message in the report panel. That's not causing it.

raindropsfromsky, 2025-02-17:
[image]

APEbbers, 2025-02-17:
git switch Develop?

raindropsfromsky, 2025-02-17:
oops. Since I am not a coder, I just copied what you recommended.

[image]

raindropsfromsky, 2025-02-17:
Yes, now the behavior is perfect! 👍 

Can you trigger menus when we hover on the **Menu** button? 🤞🏽

APEbbers, 2025-02-17:
Perhaps, i can make it an option. But it should be applied to all menus and dropdown buttons. otherwise it is inconsistent.
Please note that is not standard behaviour. Check Windws, Office, SolidWorks, SolidEdge for example. They all need a click.

kind regards

raindropsfromsky, 2025-02-17:
Yes, but we should always aim at what is more user-friendly; not what others have implemented.

All the products you mention have a long history, and they may have maintained continuity by not changing the UI drastically.
In fact, most of the large CAD customers do not prefer rapid changes in the UI, because they fear productivity loss.

So we have to look at disrupter products, which do not have legacy baggage.

APEbbers, 2025-02-17:
I'm not sure it is that user-friendly. Also, Windows is an OS which aimed to be user-friendly. I'm a mechanical engineer myself and has used a lot of different cad programs. I personally don't want the menus to be open on hover, because it will be annoying when you want to click on a button close to it.

I believe the current behaviour in Windows, Office, the major CAD programs are actually thought through than only legacy behaviour.

closed 2025-03-12 by APEbbers