APEbbers/FreeCAD-Ribbon #322: [BUG]Separators not working.

dgrabulosa, 2025-02-18:
I can't remove/disable separators.

[image]

Ribbon 1.7.3.5

`OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40303 (Git) Conda
Build type: Release
Branch: main
Hash: 145af5cddc9774f00fd4dfb3bd35e94efc5c70ca
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: Spanish/Spain (es_ES)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/141.951
Installed mods: 
  * FreeCAD-Ribbon 1.7.3.5
  * FreeCAD-Ribbon.backup1736985501.3929608 1.6.2 (Disabled)
  * lattice2 1.0.0
  * PieMenu 1.11.2
  * sheetmetal 0.7.11
`

APEbbers, 2025-02-19:
Hi @dgrabulosa ,

Are you able to export your layout and share it with me? I'll check if there any issues there that are causing this.

kind regards,

APEbbers, 2025-02-19:
Hi @dgrabulosa,

I've updated the Ribbon to version 1.7.3.7. This should fix the double separators and when you add/remove a separator the apply button will be enabled.

I've also created a script that removes toolbars that are placed under the wrong workbench. Perhaps you can run it from within FreeCAD to see if it fixes your problem? (The original "RibbonStructure.json" will be backupped so you can revert back.

The file is: [FixRibbonStructure.zip](https://github.com/user-attachments/files/18875271/FixRibbonStructure.zip)
Extract it, and open the file in FreeCAD and press F6. In the add-on folder of the ribbon should appear a "RibbonStructure.json.bak". This is the backup file.

Restart FreeCAD and test if you can remove the separators.

It would be very helpfull if you can let me know if it works. If so, I'll add it to other scripts in the Ribbon UI.

Kind regards

closed 2025-02-22 by APEbbers

dgrabulosa, 2025-02-23:
I can remove all separators. thanks. No need the script.
But if I execute the script I can remove and add separators.
Thanks.
`