APEbbers/FreeCAD-Ribbon #325: Ribbon takes a huge time to launch (3 minutes)

raindropsfromsky, 2025-02-22:
FC takes almost 3 minutes to launch.
IINW the entire delay is caused by Ribbon UI (see the following time stamps):

00:53:19  Activating Ribbon Bar...
00:53:19  NOTICE: Addon "C:\Users\naray\AppData\Roaming\FreeCAD\Mod\FreeCAD-Ribbon.backup1739537104.9362228" disabled by presence of ADDON_DISABLED stopfile
00:56:14  FreeCAD Ribbon: https://github.com/APEbbers/FreeCAD-Ribbon

It should not be this slow!

BTW, does the second line indicate an issue, or is it because of a wrong setting?
I did not disable the Ribbon addon.

APEbbers, 2025-02-22:
Hi,

The second line, shows a backup. So something went wrong. If you start FreeCAD again, does it take this long again?
perhaps you see more if you look in the log files?

raindropsfromsky, 2025-02-22:
Hi!

I had not expected a response from you today! :)
So I updated the post (see above).

raindropsfromsky, 2025-02-22:
I closed FC and launched it again.

01:11:25  Activating Ribbon Bar...
01:11:25  NOTICE: Addon "C:\Users\naray\AppData\Roaming\FreeCAD\Mod\FreeCAD-Ribbon.backup1739537104.9362228" disabled by presence of ADDON_DISABLED stopfile
01:12:27  FreeCAD Ribbon: https://github.com/APEbbers/FreeCAD-Ribbon

APEbbers, 2025-02-22:
You have to delete the backup to get rid of the message. When was your last update of the ribbon?

APEbbers, 2025-02-22:
Couple of questions:
- Did you change your layout? Perhaps with a custom dropdown button?
- Did you install a new workbench?
- The lastest updates require a reload of the workbenches in the layout menu. Have you done that?
- Can you copy the info from the about Ribbon UI dialog?

raindropsfromsky, 2025-02-22:
> When was your last update of the ribbon?

Maybe 2 days ago. I again updated it today, just before I noticed the delayed start.
(BTW, this delay was always there. I just decided to report it today, because I happened to pay attention to the time stamps.)

raindropsfromsky, 2025-02-22:
I am unable to find my log file at C:\Users\naray\AppData\Roaming\FreeCAD.

APEbbers, 2025-02-22:
it should be here:
[image]

Did you installed a workbench two days ago? or changed your layout?

raindropsfromsky, 2025-02-22:
> * Did you change your layout? Perhaps with a custom dropdown button?

No. 

> * Did you install a new workbench?

Well, after reporting the bug today, I have installed the Sheet Metal unfolder WB.
(IINW it was an integral part of the Sheet Metal WB earlier.)

> * The lastest updates require a reload of the workbenches in the layout menu. Have you done that?
No. But this should be automated.
We cannot expect users to tweak the Ribbon UI under the hood every time there is an WB update!

BTW I just refreshed the WBs, and it is taking much longer than the three minutes we started with!
So the refreshing should also get finished much faster.

> * Can you copy the info from the about Ribbon UI dialog?

It is 1.7.3.4 now.
(This is a frustrating behavior by AM: I had changed the branch to Develop earlier. So it should stay in that branch, and keep updating versions within that branch. Instead, AM resets the branch to Main in every session, and even finds the latest versions in the Main branch only. So I have to keep changing the branch and then pass through manual update with every session!)

raindropsfromsky, 2025-02-22:
I do not have a file named **freecad.log** at all.
I checked with Everything (Windows indexed search).

raindropsfromsky, 2025-02-22:
I have very fewer files compared to your listing:

[image]

raindropsfromsky, 2025-02-22:
I changed the branch to Develop, and restarted FC.
Now the Ribbon branch is 1.8.x.

When the FC started, it shows these messages in the Report panel:
01:43:13  Activating Ribbon Bar...
01:43:13  NOTICE: Addon "C:\Users\naray\AppData\Roaming\FreeCAD\Mod\FreeCAD-Ribbon.backup1739537104.9362228" disabled by presence of ADDON_DISABLED stopfile
01:44:38  FreeCAD Ribbon: https://github.com/APEbbers/FreeCAD-Ribbon

So it is still taking almost 85 seconds to load.

raindropsfromsky, 2025-02-22:
I let AM scan for update once again.
It reports an error:
`Invalid version specified for tag version in Addon FreeCAD Ribbon: 1.8.x `

APEbbers, 2025-02-22:
> I do not have a file named freecad.log at all.
I checked with Everything (Windows indexed search).

Neve mind, perhaps. i forgot that I had enabled it myself. You have to change the shortcut for it.
[image]

 Did you delete the folder with the backup?

> The lastest updates require a reload of the workbenches in the layout menu. Have you done that?
> No.` But this should be automated.
> We cannot expect users to tweak the Ribbon UI under the hood every [...]

You get a message when an reload is needed. Unfortunally, reloading cannot get any faster, because all workbenches needs to be activated in order to create a data file. The datafile is needed to load commands when an workbench is not activated.
The latest versions should be a little fasteer but not much.

> It is 1.7.3.4 now.
> (This is a frustrating behavior by AM: I had changed the branch to Develop earlier. So it should stay in that branch, and keep updating versions within that branch. Instead, AM [...]

Yes. That's why i prefer the commandline. Or after an update check the branch and switch to develop before restart.

> I changed the branch to Develop, and restarted FC.
> Now the Ribbon branch is 1.8.x.
>
> When the FC started, it shows these messages in the Report panel:
> 01:43:13 Activating Ribbon Bar...
> 01:43:13 [...]

Delete the folder and see if it helps. or move it to the desktop. You have not set any workbench to load on startup?

> I let AM scan for update once again.
> It reports an error:
> Invalid version specified for tag version in Addon FreeCAD Ribbon: 1.8.x 

as i mentioned before, just ignore this. It is just an warning. It won't affect anything.

APEbbers, 2025-02-22:
It can also help if you temporarily move the system.cfg and user.cfg to you desktop.
Sometimes messing around with settings can also cause a slow start.

If it doesn't help, you can place them back.

[image]

APEbbers, 2025-02-22:
Another idea: in the develop branch you can run:
[image]

click on it. it opens in FreeCAD. press ctrl+F6 to run (or Menu->Macro->Execute macro). 

it cleans up the RibbonStructure.json. It wil create a RibbonStructure.json.bak as backup in case you want to revert back.
When you see the backup file, you can restart FreeCAD.

raindropsfromsky, 2025-02-23:
> You get a message when an reload is needed. Unfortunally, reloading cannot get any faster, because all workbenches needs to be activated in order to create a data file. The datafile is needed to load commands when an workbench is not activated.

In that case, can you make this a background thread that runs whenever the CPU load is less?
Currently, the refresh process locks FC for an unknown period (the FC window shows "not responding..."), which is rather frustrating.

It can become more bearable if there is a progress bar, and if there is a count-down timer.
(It does not matter if the progress bar and timer are not accurate.

> That's why i prefer the commandline. Or after an update check the branch and switch to develop before restart.

Well, that's a -step manual task, which is why it is even worse than changing the branch in AM.
When we already have a sophisticated UI, why should the user have to use the command line?

raindropsfromsky, 2025-02-23:
> You have not set any workbench to load on startup?

I found that as many as 16 WBs were set to be autoloaded!
(I did not do that myself: It had happened somehow.)

I have reduced them to 6 now.

APEbbers, 2025-02-23:
Great! Good that it is solved. I'm closing the issue.

kind regards

closed 2025-02-23 by APEbbers

APEbbers, 2025-02-23:
Hi,

> ou get a message when an reload is needed. Unfortunally, reloading cannot get any faster, because all workbenches needs to be activated in order to create a data file. The datafile is needed to load [...]

I realized that the searchbar has such an indicator. I've now adapted this for the Ribbon. It is in the latest develop version.

kind regards,

raindropsfromsky, 2025-02-23:
> I realized that the searchbar has such an indicator. I've now adapted this for the Ribbon. It is in the latest develop version.

That's great!

BTW, the progress bar is provided in AM itself. 
(It's a different matter that it does not work properly. I have raised a [bug](https://github.com/FreeCAD/FreeCAD/issues/18867) for that.)

raindropsfromsky, 2025-02-23:
Did you change the datafile format?
[image]

This is going to give grief to users because every time we start FC, it will refer to the main branch, which uses the old format.
When we change the branch to Develop, we will have to update the datafile format.

APEbbers, 2025-02-23:
Well, it is the develop branch after all. For the next release or if you want to work with the develop branch, you need to update the datafile. So it is best to stay on or the main branch or the develop branch and not switch inbetween.

raindropsfromsky, 2025-02-23:
I know!

I like to stay on the Develop branch only, but the AM forces me back to the Main branch every time I start FC.
Anyhow, this is not in our hands! 🤷

raindropsfromsky, 2025-02-23:
> I realized that the searchbar has such an indicator. I've now adapted this for the Ribbon. It is in the latest develop version.

I tried to trigger this, but could not see it.
Can you please show a screenshot?
How do we get to see it?

APEbbers, 2025-02-23:
It looks like this after pressing "Reload" in the layout dialog:

[image]

Check if it went behind FreeCAD

raindropsfromsky, 2025-02-23:
> Check if it went behind FreeCAD

When I click on the **Refresh** button, the mouse icon turns into an hourglass, and remains like that until the process finishes.
There is no other display at all.
Initially, the FC window looks normal, but if I do anything in it, it turns all-white.

I checked the Desktop with a three-finger drag method. It does not show any other dialog.
[image]

APEbbers, 2025-02-23:
Can you copy this info?

[image]

raindropsfromsky, 2025-02-23:
I discovered that the Copy to clipboard button does not work (The clipboard retails its older content)

So I had to take a screenshot.
[image]

raindropsfromsky, 2025-02-23:
**Update:** 
Finally I gave up struggling with AM and tried the command line method.
My version was 2 commits behind.
I pulled Git, and now the **Copy to clipboard** button works.

But my screen looks different: It does not have the "Credits" section.

Installed version: 1.8.x
Branch: Develop
Commit ID: 831adce8648c992ddfb3cddf4fdd3660d63c140e

APEbbers, 2025-02-23:
Your on a an older version. Try git pull from command line

APEbbers, 2025-02-23:
That's because you need to have git installed in python. You will get it when the new release is out or when you reinstall the ribbon.

It is not that important. I would let it be.

raindropsfromsky, 2025-02-23:
I tried the Refresh button with the latest version of Ribbon.

I see a small window with the latest status. 👍

I could also see the status in the Reports Panel, which was even better! 👍👍
(Did I miss that in the earlier versions? Or is this a new feature?)

I'd suggest two small improvements:
1. The dialog vanishes intermittently. It should remain on top of the FC window at all times.
    (In any case, FC does not let us carry on with other operations!)
2. After the refresh cycle is over, show a final summary status message 
    (e.g. "31 Workbenches were refreshed in 249 seconds.")

raindropsfromsky, 2025-02-23:
Probably I have suggested this earlier, but can you gray out the **Refresh** button after we have finished the refresh cycle?
That way, the user does not get confused why the button is still prompting him to refresh.

If possible, the button should remain gray until FC detects a mismatch and activates the button.

APEbbers, 2025-02-23:
> I could also see the status in the Reports Panel, which was even better! 👍👍
> (Did I miss that in the earlier versions? Or is this a new feature?)

This is part of the new feature indeed.

> The dialog vanishes intermittently. It should remain on top of the FC window at all times.
> (In any case, FC does not let us carry on with other operations!)

Currently this would result in at top over all applications. Qt has not a proper way to let dialogs stay only in front of their application.

> After the refresh cycle is over, show a final summary status message
> (e.g. "31 Workbenches were refreshed in 249 seconds.")

I'm not seeing a benefit for this. Afterwards, the layout dialog is loaded anyway. But I can make a note on the layout dialog with the last time that the workbenches have been reloaded.

> Probably I have suggested this earlier, but can you gray out the Refresh button after we have finished the refresh cycle?
> That way, the user does not get confused why the button is still prompting [...]

I rather do not want that. If a user has any problems. I want the user to be able to refresh the workbenches at any given time.
The button is not prompting. It is a refresh button like Freecad's own refresh button but then for the datafile.

raindropsfromsky, 2025-02-23:
> Currently this would result in at top over all applications. Qt has not a proper way to let dialogs stay only in front of their application.

As I mentioned in the other issue, I'd like the dialog to be "on the top of all apps".
(That was my changed opinion, after comparing the options.)

I realized that the user would not be switching between apps when he is working on FC.
So if a dialog remains on top of all windows for a small interval, no big deal.