APEbbers/FreeCAD-Ribbon #327: Allow user to control the size of the help, overlay, and Preference buttons; and also the "search tools" field.

raindropsfromsky, 2025-02-23:
IINW there is no control to let the user change the size of these buttons.

[image]

APEbbers, 2025-02-23:
Yes, those are set with the tabbar size. They won't have separate settings.
The icons are retrieved from FreeCAD, but there seems a size limit on them. I'll check.
When you set the tabbar size to 16, you will see that they are scaling as wel

raindropsfromsky, 2025-02-23:
> When you set the tabbar size to 16, you will see that they are scaling as wel

I set the tabbar size to 34 and then to 30. 
In both cases, the entire toolbar looks good! 👍 

However, the text size for the **search tools, prefs...** field remains too tiny.
How do I change it?
Does it inherit its font type+size from some other control (maybe "menu"?)

Secondly, is there any way to expand this field by dragging its left border?

APEbbers, 2025-02-23:
In the develop branch are new settings to set the font size. Maybe you can check these out.? I think you need to tweak Font for buttons and menu

raindropsfromsky, 2025-02-23:
Ok I will check this out.

What about expanding  the **search tools, prefs...** field by dragging its left border?

APEbbers, 2025-02-23:
Thanks, I'll fix the typo.
The width is currently not changeable. I have to look into that, but it won't have a priority.

closed 2025-03-12 by APEbbers