APEbbers/FreeCAD-Ribbon #330: The tabbar has moved down, leaving a large blank space behind!

raindropsfromsky, 2025-02-25:
In the new development branch, the tab bar has moved down, leaving empty space behind.
[image]

As a result, a large space in the ribbon area is wasted.

**Desired:** Move it back to where it was.

APEbbers, 2025-02-25:
Hi, 

This is not a bug, but a new look. This is places the quickaccess  and right toolbar above the tabbar. This makes the ribbon more inline with MS Office and other programs with a ribbon.

You can change is back to the old layout in preferences.

kind regards

closed 2025-02-25 by APEbbers

raindropsfromsky, 2025-02-25:
Is it possible to make use of the spare space available in the Windows bar, then?

APEbbers, 2025-02-25:
No, the windows bar is fixed, I cannot change that. However, I'm thinking to show the file with path there.

raindropsfromsky, 2025-02-25:
> I'm thinking to show the file with path there.

Well, showing the path will clutter the display unnecessarily.
The main name is already displayed there (and an asterisk if the file is unsaved), which is sufficient, IMHO.

APEbbers, 2025-03-02:
Hi,

I have to come back on this:

> Is it possible to make use of the spare space available in the Windows bar, then?  
> No, the windows bar is fixed, I cannot change that. However, I'm thinking to show the file with path there.

I thought that is should actually be possible, so I looked into this. I have hidden the title bar and added the close button to the ribbon. Also I added "FreeCAD 1.1.0" to it. I think it looks more and more like a proper ribbon. 

[image]

All changes can be disabled/changed in the preference menu. In case you want to go back to the old layout.

On windows there is a bug with resizing at the moment. That's why you don't see a resize button at the moment. It has to do with MS own functionality in Windows 10 and 11. You can however drag the window and double click on the title to set it full screen.

kind regards