APEbbers/FreeCAD-Ribbon #332: Ribbon UI does not appear. Error reported.

raindropsfromsky, 2025-02-26:
00:39:57  During initialization the error "Internal C++ object (PySide2.QtWidgets.QCommonStyle) already deleted." occurred in C:\Users\naray\AppData\Roaming\FreeCAD\Mod\FreeCAD-Ribbon\.\InitGui.py
00:39:57  Please look into the log file for further information
00:39:57  During initialization the error "No module named 'PySide6'" occurred in C:\Users\naray\AppData\Roaming\FreeCAD\Mod\SearchBar\.\InitGui.py
00:39:57  Please look into the log file for further information
00:40:21  Invalid version specified for tag version in Addon FreeCAD Ribbon: 1.8.x
00:40:21  libpng warning: iCCP: duplicate
00:40:21  libpng warning: iCCP: duplicate
00:40:21  libpng warning: iCCP: duplicate
00:40:31  Invalid version specified for tag version in Addon FreeCAD Ribbon: 1.8.x
00:40:40  Traceback (most recent call last):
00:40:40    File "C:\Program Files\Util\FreeCAD\Mod\AddonManager\addonmanager_utilities.py", line 494, in run_interruptable_subprocess
00:40:40      stdout, stderr = p.communicate(timeout=1)
00:40:40                       ^^^^^^^^^^^^^^^^^^^^^^^^
00:40:40    File "C:\Program Files\Util\FreeCAD\bin\Lib\subprocess.py", line 1209, in communicate
00:40:40      stdout, stderr = self._communicate(input, endtime, timeout)
00:40:40                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
00:40:40    File "C:\Program Files\Util\FreeCAD\bin\Lib\subprocess.py", line 1630, in _communicate
00:40:40      raise TimeoutExpired(self.args, orig_timeout)
00:40:40  subprocess.TimeoutExpired: Command '['C:\\Program Files\\Util\\Git\\cmd\\git.EXE', 'fetch']' timed out after 1 seconds
00:40:40  
00:40:40  The above exception was the direct cause of the following exception:
00:40:40  
00:40:40  Traceback (most recent call last):
00:40:40    File "C:\Program Files\Util\FreeCAD\Mod\AddonManager\addonmanager_workers_startup.py", line 561, in run
00:40:40      checker.check_package(repo)

-----------------

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40421 (Git) Conda
Build type: Release
Branch: main
Hash: f9bb6508ffb420502d7b91592d87ea4f95c399d4
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical/physical DPI: 96/123.928
Installed mods: 
  * Beltrami 1.1.0
  * CommandPanel
  * CurvedShapes 1.0.13
  * Curves 0.6.58
  * ExplodedAssembly
  * fasteners 0.5.37
  * frame 0.1.1
  * FrameForge 0.1.2
  * FreeCAD-Ribbon 1.8.0
  * freecad.gears 1.3.0
  * FreeGrid 2.2.0
  * OpenTheme 2024.9.1
  * parts_library
  * Quetzal 1.3.3
  * SearchBar 1.4.0
  * sheetmetal 0.7.11
  * Silk 0.1.5
  * symbols_library

00:40:40    File "C:\Program Files\Util\FreeCAD\Mod\AddonManager\addonmanager_workers_startup.py", line 657, in check_package
00:40:40      self.check_workbench(package)
00:40:40    File "C:\Program Files\Util\FreeCAD\Mod\AddonManager\addonmanager_workers_startup.py", line 613, in check_workbench
00:40:40      if self.git_manager.update_available(clonedir):
00:40:40         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
00:40:40    File "C:\Program Files\Util\FreeCAD\Mod\AddonManager\addonmanager_git.py", line 211, in update_available
00:40:40      self._synchronous_call_git(["fetch"])
00:40:40    File "C:\Program Files\Util\FreeCAD\Mod\AddonManager\addonmanager_git.py", line 455, in _synchronous_call_git
00:40:40      proc = utils.run_interruptable_subprocess(final_args)
00:40:40             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
00:40:40    File "C:\Program Files\Util\FreeCAD\Mod\AddonManager\addonmanager_utilities.py", line 502, in run_interruptable_subprocess
00:40:40      raise ProcessInterrupted() from timeout_exception
00:40:40  addonmanager_utilities.ProcessInterrupted

APEbbers, 2025-02-26:
Hi,

I was working on something. Can you do a git pull and see if it works again?

Kind regards

raindropsfromsky, 2025-02-27:
While updating, I found something odd: At first, the switch command said that I was already in Develop branch _and_ up to date.
But then when I pulled, it went ahead and updated!
[image]

A git pull on Searchbar got me this:
[image]

But after all this, when I started FC, I am still unable to see Ribbon UI.
I see the following errors:
```
09:30:05  During initialization the error "Internal C++ object (PySide2.QtWidgets.QCommonStyle) already deleted." occurred in C:\Users\naray\AppData\Roaming\FreeCAD\Mod\FreeCAD-Ribbon\.\InitGui.py
09:30:05  Please look into the log file for further information
09:30:32  Previous cache process was interrupted, restarting...
09:30:33  Invalid version specified for tag version in Addon FreeCAD Ribbon: 1.8.x
09:30:44  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
09:30:44  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
09:30:45  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
09:30:45  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
09:30:45  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
09:30:46  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
09:30:46  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
09:30:46  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
09:30:46  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
09:30:46  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
09:30:47  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
09:30:47  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
09:30:47  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
09:30:47  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
09:30:48  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
09:30:48  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
09:30:48  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
09:30:48  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
09:30:49  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
09:30:49  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
09:30:49  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
09:30:49  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
09:30:49  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
09:30:50  libpng warning: iCCP: duplicate
09:30:50  libpng warning: iCCP: duplicate
09:30:56  Invalid version specified for tag version in Addon FreeCAD Ribbon: 1.8.x
09:31:08  libpng warning: iCCP: duplicate
```

raindropsfromsky, 2025-02-27:
The second line states:
`Please look into the log file for further information`

Where is this log file? Will you need it for troubleshooting?

APEbbers, 2025-02-27:
It would be helpfull!! If you add  --write-log to your FreeCAD shortcut like this:
[image]

After starting FreeCAD, in you will get this:
[image]

The log will point to the exact file and line were the error occurs.

raindropsfromsky, 2025-02-27:
I modified the properties of shortcut in Taskbar.

This error is displayed on launch:
```
17:38:06  During initialization the error "Internal C++ object (PySide2.QtWidgets.QCommonStyle) already deleted." occurred in C:\Users\naray\AppData\Roaming\FreeCAD\Mod\FreeCAD-Ribbon\.\InitGui.py
17:38:06  Please look into the log file for further information
```

[FreeCAD.log](https://github.com/user-attachments/files/19008630/FreeCAD.log)

APEbbers, 2025-02-27:
Hi,

Thanks the logfile helped a lot. Did you set the stylesheet to none by any change? The function that caused the error, was only used when no stylesheet was set. Anyway, I've modified the code. Can you try with the latest version of Develop?

Kind  regards

raindropsfromsky, 2025-02-27:
No, I haven't touched the stylesheet at all.

I again faced the same weird problem:
[image]

I am able to see the Ribbon now. Thanks!

APEbbers, 2025-02-27:
No problem!. I see that you do and a "git switch" and a 'git pull". The message that your already up to date comes from the 'git switch". It is actually saying that you are already on the develop branch.

Once you are on the develop branch, "git pull" is enough. You don't need to do a "git switch". That is only needed when you want to switch to another branch.

Only the AM switches back to the main branch every time. From the commandline, you will stay on the same branch.

kind regards

closed 2025-02-27 by APEbbers

raindropsfromsky, 2025-02-27:
> Only the AM switches back to the main branch every time. From the commandline, you will stay on the same branch.

Ah! That was the root of my bad habit!

But the issue still remains, IMHO.

It is ok if Git tells me that I am already on the Develop branch.
But why does it say that 'your branch is up to date..."?
Does it not refer to the tip version in that branch?

In other words, I expected something like "_you are x commits behind_."

APEbbers, 2025-02-27:
It is a strange message indeed. Especially when there are updates on the develop branch. But i've accepted the strange messages that you get sometimes. If I'm not sure, I do an extra git pull to check