APEbbers/FreeCAD-Ribbon #338: Feature request: Please provide more space (margins) to the WB refresh dialog.

raindropsfromsky, 2025-03-02:
During WB list refresh, the dialog box looks too cramped. 
[image]

Please make it more spacious.

raindropsfromsky, 2025-03-06:
Is the dialog box removed altogether?
I cannot see it anymore!

Here's the log in **Report** panel:

```
08:39:27  Activating Ribbon Bar...
08:39:27  FreeCAD loaded without titlebar
08:39:47  FreeCAD Ribbon: https://github.com/APEbbers/FreeCAD-Ribbon
08:39:58  Previous cache process was interrupted, restarting...
08:39:59  Invalid version specified for tag version in Addon FreeCAD Ribbon: 1.8.x
08:40:07  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
08:40:07  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
08:40:07  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
08:40:07  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
08:40:08  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
08:40:08  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
08:40:08  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
08:40:08  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
08:40:09  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
08:40:09  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
08:40:09  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
08:40:09  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
08:40:09  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
08:40:10  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
08:40:10  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
08:40:10  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
08:40:10  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
08:40:11  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
08:40:11  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
08:40:11  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
08:40:12  Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError 
08:40:12  libpng warning: iCCP: duplicate
08:40:12  libpng warning: iCCP: duplicate
08:40:19  Invalid version specified for tag version in Addon FreeCAD Ribbon: 1.8.x
08:40:30  libpng warning: iCCP: duplicate
```

APEbbers, 2025-03-06:
Hi,

I just tested it on Windows, Linux Mint and MacOS, and it is working fine. Can you send me your log file and the info from the about ribbon dialog?

Kind regards,

raindropsfromsky, 2025-03-06:
oops! Terrible mixup: I actually launched the WB update check by Addon Manager! 
(I have set the AM to look for updates when I launch AM.)

I confirmed just now that the Refresh operation works well.
Even the message at the end has become more user-friendly.

closed 2025-03-06 by raindropsfromsky

APEbbers, 2025-03-06:
aha, okay can happen no worries.! Glad to here that this issue is fixed.

Kind regards,