APEbbers/FreeCAD-Ribbon #340: Instead of "menu" button, please provide the hamburger icon

raindropsfromsky, 2025-03-02:
Currently, the menu system is triggered from a "text-only" button.
[image]

Instead, 
1. Separate the FC logo. 
    Place it in the left-top corner. 
    If the user clicks on it, it should show the usual Windows menu.
    [image]

2. Instead of "Menu" button, show the **hamburger** icon. 
    [image]
    [image]

APEbbers, 2025-03-04:
Hi,

I'm not going to implement this. The reason is that FreeCAD is not only used by windows but also by Linux and MacOS. These don't have a menu like this. Also to have a prominent button like this is on purpose.  I believe that it makes the function of the menubutton more clear and easier to find.  The "hamburger" icon is more often used for smartphone/tablet apps and websites. 

I might bring back the circular button back instead.

Kind regards

closed 2025-03-04 by APEbbers

raindropsfromsky, 2025-03-05:
> The reason is that FreeCAD is not only used by windows but also by Linux and MacOS.

👍