APEbbers/FreeCAD-Ribbon #343: The latest version of Ribbon caused an endless series of critical errors on launch. FC does not start at all!

raindropsfromsky, 2025-03-07:
Today I did a Git pull, and started FC.
 
Even before FC showed its flash screen, it popped up a message that this is based on old format. Do you want to switch to new format?

I selected yes, and then it showed this dialog:
[image]
(Cannot read what it says, as it shows black text on black background)

After that, it lapsed in an endless loop of this error:
[image]

I kept on closing the error, but it keeps popping up again and again.

FC does not start at all.

APEbbers, 2025-03-07:
I'm still working on it. You can try the last version or switch to the "Pre-release-v1.8.0" branch for now.
I'm doing tests and such, so that branch will be more stable.

Kind regards

closed 2025-03-07 by APEbbers

raindropsfromsky, 2025-03-07:
Yes, it works. Thanks!

raindropsfromsky, 2025-03-10:
I am seeing the message on the "Pre-release-v1.8.0" branch also.
[image]
But I can press ESC to dismiss it, and carry on.