APEbbers/FreeCAD-Ribbon #348: The customized panel layout is lost when I close FC. This may be due to the faulty initialization of the Ribbon WB

raindropsfromsky, 2025-03-11:
I used command line mode, switched to the **Develop** branch, and used the Git pull command.

Each time I start FC, it launches the same dialog: **[BUG#1]**
[image]

If I select **Yes**, It simply flashes a small dialog (one line high, about 8 cm wide).
Even before this dialog is fully visible, it disappears. 
So I have never actually seen what it is, but it looks like the "Refresh WB list" dialog of the Ribbon WB.

At the same time, FC also shows the flash screen.
But again, it appears only for an instant, and disappears.

Within a second of the launch, the FC screen is presented.
There is no wait time at all.

In other words, except for flashing two rectangles, the FC does not make us wait.
It simply starts up almost instantly. 
**This indicates that something has gone wrong.**
(Earlier, it used to take about 90 seconds, as I have set a couple of WBs to be pre-loaded.)

When the FC starts, the screen layout is reset to a default state, as shown below: **[BUG#2]**
[image]

I prefer to position four panels as shown below:
[image]
So each time I launch FC, I have to manually launch the panels I need, and move them around.
This is too tedious.

**Probably the faulty initialization of the Ribbon deletes the saved layout.**

-----

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical/physical DPI: 96/123.928
Installed mods: 
  * AnimationFreeCAD 1.0.0
  * Beltrami 1.1.0
  * CommandPanel
  * CurvedShapes 1.0.13
  * Curves 0.6.59
  * ExplodedAssembly
  * fasteners 0.5.37
  * frame 0.1.1
  * FrameForge 0.1.2
  * FreeCAD-Ribbon 1.8.0
  * freecad.gears 1.3.0
  * FreeGrid 2.2.0
  * OpenTheme 2024.9.1
  * parts_library
  * Quetzal 1.3.3
  * SearchBar 1.4.1
  * sheetmetal 0.7.21
  * Silk 0.1.5
  * symbols_library

raindropsfromsky, 2025-03-11:
Here is a screen capture of this problem:

[image]

APEbbers, 2025-03-12:
Hi,

I found the issue and fixed it. It was caused by the message blocking the startup of FreeCAD. Now FreeCAD is not blocked when the message shows and continues starting.

kind regards

raindropsfromsky, 2025-03-12:
I tested both **Develop** and **pre-release** branches. In both versions, FC cannot start properly.

When I launch FC, the following error pops up in both branches:
[image]

Then the FC window appears in a weird form:
[image]
It does not have Windows controls. It cannot be resized or maximized.

APEbbers, 2025-03-12:
Sorry, should be fixed now!

raindropsfromsky, 2025-03-12:
> Can you test if the problem still occurs when the command panel addon is disabled?

I removed the **Command** panel addon.
(To my surprise, it turned out to be an outdated addon! I have never used it, and I didn't like it, either!)

Now the behavior looks ok even in the **Main** branch, and also the **Pre-release** branch. 👍

closed 2025-03-12 by raindropsfromsky