APEbbers/FreeCAD-Ribbon #351: v1.8.0 error

HakanSeven12, 2025-03-13:
It crushed after update. I tried a clean install also but result is same.

OS: Windows 10 build 19044
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Turkish/Turkey (tr_TR)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * FreeCAD-Ribbon 1.8.0
  * Road 2025.2.22

 
08:35:35  Activating Ribbon Bar...
08:35:37  Ribbon UI: Custom text are reset because the language was changed
08:35:37  FreeCAD Ribbon: https://github.com/APEbbers/FreeCAD-Ribbon
08:35:37  Uncaught exception:
   File "C:\Users\hakan.seven\AppData\Roaming\FreeCAD\Mod\FreeCAD-Ribbon\.\FCBinding.py", line 3244, in __init__
    ribbon = ModernMenu()
             ^^^^^^^^^^^^
  File "C:\Users\hakan.seven\AppData\Roaming\FreeCAD\Mod\FreeCAD-Ribbon\.\FCBinding.py", line 454, in __init__
    LatestVersion = StandardFunctions.ReturnXML_Value_Git(User, Repo, Branch, File, ElementName)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\hakan.seven\AppData\Roaming\FreeCAD\Mod\FreeCAD-Ribbon\.\Standard_Functions_RIbbon.py", line 472, in ReturnXML_Value_Git
    response = requests.get(url)
               ^^^^^^^^^^^^^^^^^
  File "C:\Users\hakan.seven\AppData\Roaming\FreeCAD\Mod\FreeCAD-Ribbon\.\Resources\packages\requests_local\api.py", line 73, in get
    return request("get", url, params=params, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\hakan.seven\AppData\Roaming\FreeCAD\Mod\FreeCAD-Ribbon\.\Resources\packages\requests_local\api.py", line 59, in request
    return session.request(method=method, url=url, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\hakan.seven\AppData\Roaming\FreeCAD\Mod\FreeCAD-Ribbon\.\Resources\packages\requests_local\sessions.py", line 590, in request
    resp = self.send(prep, **send_kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\hakan.seven\AppData\Roaming\FreeCAD\Mod\FreeCAD-Ribbon\.\Resources\packages\requests_local\sessions.py", line 704, in send
    r = adapter.send(request, **kwargs)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\hakan.seven\AppData\Roaming\FreeCAD\Mod\FreeCAD-Ribbon\.\Resources\packages\requests_local\adapters.py", line 698, in send
    raise SSLError(e, request=request)

SSLError: HTTPSConnectionPool(host='raw.githubusercontent.com', port=443): Max retries exceeded with url: /APEbbers/FreeCAD-Ribbon/main/package.xml (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1006)')))
Traceback (most recent call last):
  File "C:\Users\hakan.seven\Documents\FreeCAD 1.0\bin\Lib\site-packages\urllib3\connectionpool.py", line 466, in _make_request
    self._validate_conn(conn)
  File "C:\Users\hakan.seven\Documents\FreeCAD 1.0\bin\Lib\site-packages\urllib3\connectionpool.py", line 1095, in _validate_conn
    conn.connect()
  File "C:\Users\hakan.seven\Documents\FreeCAD 1.0\bin\Lib\site-packages\urllib3\connection.py", line 730, in connect
    sock_and_verified = _ssl_wrap_socket_and_match_hostname(
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\hakan.seven\Documents\FreeCAD 1.0\bin\Lib\site-packages\urllib3\connection.py", line 909, in _ssl_wrap_socket_and_match_hostname
    ssl_sock = ssl_wrap_socket(
               ^^^^^^^^^^^^^^^^
  File "C:\Users\hakan.seven\Documents\FreeCAD 1.0\bin\Lib\site-packages\urllib3\util\ssl_.py", line 469, in ssl_wrap_socket
    ssl_sock = _ssl_wrap_socket_impl(sock, context, tls_in_tls, server_hostname)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\hakan.seven\Documents\FreeCAD 1.0\bin\Lib\site-packages\urllib3\util\ssl_.py", line 513, in _ssl_wrap_socket_impl
    return ssl_context.wrap_socket(sock, server_hostname=server_hostname)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\hakan.seven\Documents\FreeCAD 1.0\bin\Lib\ssl.py", line 502, in wrap_socket
    return self.sslsocket_class._create(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\hakan.seven\Documents\FreeCAD 1.0\bin\Lib\ssl.py", line 1048, in _create
    self.do_handshake()
  File "C:\Users\hakan.seven\Documents\FreeCAD 1.0\bin\Lib\ssl.py", line 1317, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1006)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\hakan.seven\Documents\FreeCAD 1.0\bin\Lib\site-packages\urllib3\connectionpool.py", line 789, in urlopen
    response = self._make_request(
               ^^^^^^^^^^^^^^^^^^^
  File "C:\Users\hakan.seven\Documents\FreeCAD 1.0\bin\Lib\site-packages\urllib3\connectionpool.py", line 490, in _make_request
    raise new_e
urllib3.exceptions.SSLError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1006)

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Users\hakan.seven\AppData\Roaming\FreeCAD\Mod\FreeCAD-Ribbon\.\Resources\packages\requests_local\adapters.py", line 667, in send
    resp = conn.urlopen(
           ^^^^^^^^^^^^^
  File "C:\Users\hakan.seven\Documents\FreeCAD 1.0\bin\Lib\site-packages\urllib3\connectionpool.py", line 843, in urlopen
    retries = retries.increment(
              ^^^^^^^^^^^^^^^^^^
  File "C:\Users\hakan.seven\Documents\FreeCAD 1.0\bin\Lib\site-packages\urllib3\util\retry.py", line 519, in increment
    raise MaxRetryError(_pool, url, reason) from reason  # type: ignore[arg-type]
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='raw.githubusercontent.com', port=443): Max retries exceeded with url: /APEbbers/FreeCAD-Ribbon/main/package.xml (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1006)')))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\hakan.seven\AppData\Roaming\FreeCAD\Mod\FreeCAD-Ribbon\.\FCBinding.py", line 3244, in __init__
    ribbon = ModernMenu()
             ^^^^^^^^^^^^
  File "C:\Users\hakan.seven\AppData\Roaming\FreeCAD\Mod\FreeCAD-Ribbon\.\FCBinding.py", line 454, in __init__
    LatestVersion = StandardFunctions.ReturnXML_Value_Git(User, Repo, Branch, File, ElementName)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\hakan.seven\AppData\Roaming\FreeCAD\Mod\FreeCAD-Ribbon\.\Standard_Functions_RIbbon.py", line 472, in ReturnXML_Value_Git
    response = requests.get(url)
               ^^^^^^^^^^^^^^^^^
  File "C:\Users\hakan.seven\AppData\Roaming\FreeCAD\Mod\FreeCAD-Ribbon\.\Resources\packages\requests_local\api.py", line 73, in get
    return request("get", url, params=params, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\hakan.seven\AppData\Roaming\FreeCAD\Mod\FreeCAD-Ribbon\.\Resources\packages\requests_local\api.py", line 59, in request
    return session.request(method=method, url=url, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\hakan.seven\AppData\Roaming\FreeCAD\Mod\FreeCAD-Ribbon\.\Resources\packages\requests_local\sessions.py", line 590, in request
    resp = self.send(prep, **send_kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\hakan.seven\AppData\Roaming\FreeCAD\Mod\FreeCAD-Ribbon\.\Resources\packages\requests_local\sessions.py", line 704, in send
    r = adapter.send(request, **kwargs)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\hakan.seven\AppData\Roaming\FreeCAD\Mod\FreeCAD-Ribbon\.\Resources\packages\requests_local\adapters.py", line 698, in send
    raise SSLError(e, request=request)
requests_local.exceptions.SSLError: HTTPSConnectionPool(host='raw.githubusercontent.com', port=443): Max retries exceeded with url: /APEbbers/FreeCAD-Ribbon/main/package.xml (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1006)')))

HakanSeven12, 2025-03-13:
Looks like it tries to reach internet but Im using a company computer and there is limitations for internet access. Maybe that is the reason?

APEbbers, 2025-03-13:
Hi,

Yes I believe that's the problem indeed.  it is checking if a new version is available. I've put the line in a try-catch statement. 
The issue should be solved with version v1.8.0.1. can you check?

kind regards

HakanSeven12, 2025-03-13:
It woks! thanks. Another reminder. "Hide the titlebar of FreeCAD" is unchecked by default.

APEbbers, 2025-03-13:
Great!. I've now also set the "Hide the titlebar of FreeCAD" checked by default.

Kind regards

closed 2025-03-13 by APEbbers