APEbbers/FreeCAD-Ribbon #352: TitleBar not hidden when configured to do so.

obelisk79, 2025-03-13:
Here is my System Information. 

I am running an Arch Linux based system (EndeavourOS) using the KDE/Plasma desktop with the default Wayland compositor.

```
OS: EndeavourOS (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.1.0dev.40609 (Git)
Build date: 2025/03/09 13:41:26
Build type: Release
Branch: main
Hash: 844cceb04f0a82f55dee46445c4d57a7e20111ef
Python 3.13.2, Qt 6.8.2, Coin 4.0.3, Vtk 9.4.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/93.2319/1
Installed mods: 
  * Curves 0.6.58
  * Manipulator 1.5.7
  * ArchTextures
  * SearchBar 1.4.1
  * FreeCAD-Ribbon 1.8.0.2
  * OpenTheme 2024.9.1
  * IconThemes
  * Silk 0.1.5
  * FreeCAD-Telemetry 1.0.0
  * sheetmetal 0.7.21
```

obelisk79, 2025-03-13:
Using the following Macro has instantaneous results however:

`This macro toggles the visibility of the main window's title bar.

import FreeCAD
import FreeCADGui
from PySide6 import QtCore, QtGui, QtWidgets

def toggle_title_bar():
    """Hides or shows the title bar of the main window."""

    main_window = Gui.getMainWindow()
    current_flags = main_window.windowFlags()

    if current_flags & QtCore.Qt.FramelessWindowHint:
        # Title bar is currently hidden.  Show it.
        new_flags = current_flags & ~QtCore.Qt.FramelessWindowHint
        main_window.setWindowFlags(new_flags)
        main_window.show()  # Necessary to apply the changes immediately.
        FreeCAD.Console.PrintMessage("Title bar shown. \n")
    else:
        # Title bar is currently shown. Hide it.
        new_flags = current_flags | QtCore.Qt.FramelessWindowHint
        main_window.setWindowFlags(new_flags)
        main_window.show()
        FreeCAD.Console.PrintWarning("Title bar hidden.\n")

toggle_title_bar()`

APEbbers, 2025-03-13:
Hi,

I added the "current_flags" to the ribbon. can you test if it helps on the  branch for this issue? The link is: https://github.com/APEbbers/FreeCAD-Ribbon/tree/352-titlebar-not-hidden-when-configured-to-do-so

obelisk79, 2025-03-13:
I removed the ribbon, and copied that branch in clean. I don't see where current_flags is shown in the ribbon.

[image]

APEbbers, 2025-03-13:
The current_flags is in initGui.py there the window flags are set.
[image]

Just checking: under ribbon preferences, "Hide the titlebar" is still checked and under parameters, the key : Hide_TitleBar_FC is set to True?
[image]

obelisk79, 2025-03-13:
The parameters are set, but the behavior I'm experiencing has not changed.

obelisk79, 2025-03-13:
CustomizeWindowHint does not work. However if I modify it to use FramelessWindowHint it works as expected.

APEbbers, 2025-03-13:
Strange,  I double checked on linux mint with the main branch and no problem there.  So it has to do with the window flags and specific linux versions. 

I'm not using FramelessWindowHint because it makes the window not resizable. That's the reason I used CustomizeWindowHint.

Can it be that it has to do with the KDE Plasma dekstop? from what I read on the internet, KDE is based on Qt itself.

obelisk79, 2025-03-13:
It certainly might be related to KDE plasma, but KDE is created by the same people at Qt. From what I understand of its history, Qt started off as a framework written *for* KDE.

APEbbers, 2025-03-13:
Okay, I'm going to setup an virtal linux with KDE Plasma installed tomorrow to do some test with various windowflags to see what's going on. 

I'm come back to you on this.

APEbbers, 2025-03-14:
HI @obelisk79 ,

I've installed EndeavourOS with KDE/Plasma on virtualbox. I still have no problems, The titlebar is hidden.
However I'm using an AppImage instead of a compiled version.  (Both released and latest weekly build)

Can you try and test if the issue is gone when using an AppImage?

closed 2025-03-14 by obelisk79

APEbbers, 2025-07-27:
Hi @JaneSmith,

What kind of system are you running? I'm running EndavourOS with Wayland and have no problem hiding the titlebar.

APEbbers, 2025-08-13:
> The titlebar doesn't seem to get hidden when running FreeCAD under Wayland. It hides it fine with X.

Hi @JaneSmith,

 just found out that when you use AppImage, it works under Wayland as well. Without AppImage, it doesn´t work.
Probally a bug with QT and wayland.

kind regards