APEbbers/FreeCAD-Ribbon #353: list index out of range on v1.8.0

krom, 2025-03-14:
**Describe the bug**
`list index out of range` avter  updating plugin

**To Reproduce**
Steps to reproduce the behavior:
1. Git checkout
2. Run FreeXad
4. See error

**Expected behavior**
```shell
~/.var/app/org.freecad.FreeCAD/data/FreeCAD/Mod/FreeCAD-Ribbon on  v1.7.0 ⌚ 13:51:30
$ git checkout v1.8.0.2
Previous HEAD position was f7a271a Merge pull request #268 from APEbbers/Pre-Release-v1.7.0
HEAD is now at 99ff275 Fix for issue: Titlebar not hidden as default
```

**Screenshots**

[image]

**Please complete the following information:**
 - OS: Kubuntu 24.10
 - FreeCAD information: 
OS: KDE Flatpak runtime (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Installed mods: 
  * fasteners 0.5.37
  * boltsfc 2022.11.5
  * 3D_Printing_Tools
  * FreeCAD-Ribbon 1.7.0
  * OpenTheme 2024.9.1

- Ribbon information (Develop version only):
  v1.8.0 and v1.8.0.2
- [layout.zip](https://github.com/user-attachments/files/19246006/layout.zip)

**Additional context**
v1.7.0 works fine

krom, 2025-03-14:
```
Installed version: 1.8.0.3
Branch: main
Commit ID: c03a16ee7e717a5df8e42955ba6da17ea89aa34d
```
**And it works for me.**
Thanks.

APEbbers, 2025-03-14:
No problem. Great to hear that it is solved.

closed 2025-03-14 by APEbbers