APEbbers/FreeCAD-Ribbon #371: Cancelling "Import workbench" causes critical error

Awesome-0, 2025-05-08:
**Describe the bug**
In the Ribbon design window, in the "Import workbench layout" section, I selected Sketcher and then clicked "Import workbench" to see what it does. It opens a file explorer window. Since I have nothing to import, I click cancel (or the close window X on the top-right). This causes a critical error.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Ribbon design' (by using ALT+L)
2. Go to 'Initial setup' tab
3. Click on 'Import workbench
4. Click cancel or X in the File Explorer window that just opened
5. See error (hidden behind unclickable 'Ribbon design' window``

**Expected behavior**
I expect it to just close the file explorer window without causing an error.

**Screenshots**

[image]

[image]

[image]

**Please complete the following information:**
 - OS: Windows 10 Pro
 - FreeCAD information:
   OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * FreeCAD-Ribbon 1.9.0
  * SaveAndRestore 0.2.0
  * SearchBar 1.4.2

- Ribbon information (Develop version only):
  Installed version: 1.9.0
Branch: main
Commit ID: 182481432b9530b4ef2d0eb4c9b98f955b9f78bf
- Export your layout, zip it and add it to your issue by dragging it or use the button below.
I have not changed anything really in the layout, but here it is: 

[Export for Bug Report.zip](https://github.com/user-attachments/files/20103481/Export.for.Bug.Report.zip)

**Additional context**
While gathering the FreeCAD information the following occured: I click help icon on top-right. Drop-down menu opens. I click about, another dropdown menu opens. I click about FreeCAD. Nothing appears to happen. I try to click anything in FreeCAD, I can't (I hear the windows 'warning' sound). I hover over the freecad window on my taskbar to look behind the still open 'Ribbon design' window. Apparently the 'About' window opened behind the 'Ribbon disign' window. Now I cannot click anything to either close the 'Ribbon design' window or the 'About' window..... I have to force-close FreeCAD using task manager.

Sorry for unintentionally doing what are basically 2 bug reports in 1.

APEbbers, 2025-05-08:
Hi, 

Thanks for bringing this up. I'll fix the first bug. The second is a bit difficult, because the Ribbon design windows is set on top to avoid being opened behind FreeCAD. I look into this as well. However, in the meantime, when this happens again, this will help:
- click on FreeCAD to make sure it is focussed.
- Then press 'Window key + left or right arrow.
This moves the hidden dialog the side of the screen. Now you can drag it so, that you can click "OK". Below a screenclip:

https://github.com/user-attachments/assets/fefcfa03-aa73-4050-b074-edbc60a124cd  

I hope this helps with the second bug.

Kind regards

closed 2025-05-08 by APEbbers

Awesome-0, 2025-05-09:
Cheers! Good work :)
Also thanks for the tip with the keyboard shortcut.