APEbbers/FreeCAD-Ribbon #374: Error message since 1.9.0.1

ucool321, 2025-05-11:
.**Describe the bug**
Error message pops up at FreeCAD start. No Ribbon Menu will be loaded. Error message appear since update to v1.9.0.1

**To Reproduce**
Steps to reproduce the behavior:
1. Start FreeCAD

**Screenshots**

[image]

**Please complete the following information:**
 - OS: [e.g. iOS]
 - FreeCAD information:
  OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.13.3, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Installed mods: 
  * Gridfinity 0.11.9
  * SearchBar 1.4.2
  * OpenTheme 2024.9.1
  * FreeCAD-Ribbon 1.9.0.1

APEbbers, 2025-05-11:
Hi,

Can you try of the issue is solved with version 1.9.0.2 of the Ribbon?

kind regards

ucool321, 2025-05-11:
Problem solved. Thx for the quick support.

closed 2025-05-11 by ucool321