APEbbers/FreeCAD-Ribbon #388: Unable to control versions of Ribbon UI

raindropsfromsky, 2025-07-19:
1. When I launch the Addon Manager, it shows v 1.9.1.7 for Ribbon UI.
2. When I check the Help > About, it shows v1.7.3.4 for Ribbon UI.
3. When I select change branch in the Addon Manager, it does not show either of these two versions.

**Here's the list:**
origin/DragDrop
origin/temp
origin/352-titlebar-not-hidden-when-configured-to-do-so
Pre-release-vl .8.0
origin/LayoutProposal
main
origin/294-critical-error-on-switching-to-some-workbenches-tabs
origin/Drag-Drop-Develop
origin/Translations

This is confusing.

**Desired:**
1. Both Addon Manager and Help>About must show the correct version numbers
2. Mention version number in the change branch dialog.

raindropsfromsky, 2025-07-19:
Yes, I used to run git commands (switch and pull) on this folder:
C:\Users\<userName>\AppData\Roaming\FreeCAD\Mod

But now I cannot decide.

While the Main branch is obvious, I wanted to switch to the latest.

This branch was labeled "Development" earlier, and which has 1.9.x versions. 
But now there are too many branches (see the list), and I cannot decide which one is the latest.

The "pre-release" version is 1.8.x, which is below the 1.9.x I expected to see.

That's the issue.

APEbbers, 2025-07-19:
Your list shows also branches that do not exist anymore. I just released version 1.9.1.8, so the main and Develop branch are currently the same.

For the latest release, use main. For the latest development version, use Develop. **Don't use  the other branches.**
FYI: these are the current available branches:
[image]

closed 2025-07-20 by APEbbers