APEbbers/FreeCAD-Ribbon #397: Icons on tab have extra pixels

TomPcz, 2025-08-08:
**Describe the bug**
When tab style is set to _Icon only_, the icons have a few extra pixels on the right (white or blue, it seems to change). They almost seem like a part of a letter (Part Design - X, TechDraw - D).

**Screenshots**

[image]

[image]

 
```
OS: openSUSE Tumbleweed (KDE/plasmawayland/wayland)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/08/07 10:11:14
Build type: Release
Branch: main
Hash: 321cc5ced5456d8c4fd4ab9a042ac1bf2d143089
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: Czech/Czechia (cs_CZ)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/105.787/1
Installed mods: 
  * SaveAndRestore 0.2.1
  * BillOfMaterials 1.1.0.1
  * FreeCAD-Ribbon 1.9.1.10
  * InventorLoader 1.5.1
  * OpticsWorkbench 1.2.2
  * Render 2024.12.15
  * SearchBar 1.5.0
  * Vars 0.0.2.beta4
  * fasteners 0.5.39
  * kicadStepUpMod 11.7.3
  * sheetmetal 0.7.24
  * freecad.gears 1.3.0
  * lattice2 1.0.0
  * FusedFilamentDesign 0.25.250
  * ThreadProfile 1.96.0
```

TomPcz, 2025-08-08:
I went through the code, and it is indeed the text. It does not have the correct colour to be hidden. I am using a combination of the `FreeCAD dark` theme and the `FreeCAD` stylesheet.

[image]

[image]

APEbbers, 2025-10-05:
Hi @TomPcz ,

It has been a while, but are you able to test the Develop branch to see if the issue is resolved? Since recently, you can switch from branch in the latest version of Add-on manager.

I've made some stylesheet changes, so that the Ribbon UI is compatible with the new FreeCAD stylesheets. I also included a fix for this issue.

Kind regards

BTW, the 'experimental features' are context menus to change buttons per button. Right click on the ribbon and click 'customize'. Then you can right click on a button  to change size, type and enable text. Also I'm working on dragdrop functionality. Be aware that is still in early fases tough.